reinhard-brandstaedter/solarflow-bt-manager #3: Should the device ID be a variable?

mavo, 2024-01-31:
Hey,

as "promised" in the other ticket - here we go.
When reading through the code in order to grasp how it works I saw a very familiar ID from the example config for sf-control.
Should [this](https://github.com/reinhard-brandstaedter/solarflow-bt-manager/blob/master/src/solarflow-bt-manager.py#L115C41-L115C49) not use a variable rather then "your" sf-device id?

MaVo

reinhard-brandstaedter, 2024-01-31:
Thats basically correct, but as the bt-manager is now usually only used in a one-off call to take a hub offline the continuous reporting mode (proxy-mode) isn’t really required anymor (bt reporting is unstable).
So that code isn’t really needed/executed anymore.

mavo, 2024-01-31:
Ah cool, good to know.

Another topic:
You specify in the readme, that one needs to have basically two Wi-Fi's to get it into its offline mode.

The proper way would basically be, use my phone as a hotspot, connect sf into this wifi, let the bt-manager do its job of getting it offline and move to my main wifi?
Did I miss interpret the instructions, or how's the flow for the first time setup and using it directly with sf-control via mqtt?

reinhard-brandstaedter, 2024-02-01:
I had cases where the hub didn’t accept the update of the existing WiFi configuration. But that was only once or twice.
Now it should work without any issue, if it doesn’t on the first try, just repeat. If the script can’t connect via BT try moving closer, eventually resetting the hub into setup mode to ensure it turns on BT.
Also do not perform that switch if the hub has no power or low power.
There is nothing that could go badly wrong (or brick the hub). Last resort is a factory reset of the hub and try again.

closed 2024-02-02 by mavo