reinhard-brandstaedter/solarflow-bt-manager #6: Crashing due to dbus Error

drschnalli, 2024-02-21:
Hey there, thanks for your work! Yesterday your script was working fine after i started it for 2-3 times and it received data from my solarhub. But today when i run solarflow-bt-manager-py -i it gives me the following error message:

`2024-02-21 12:17:05,450:INFO: Found device: 94:C9:xx:xx:xx:xx: ZenP1_xx
Traceback (most recent call last):
  File "/home/drschnalli/solarflow-bt-manager/src/solarflow-bt-manager.py", line 235, in <module>
    main(sys.argv[1:])
  File "/home/drschnalli/solarflow-bt-manager/src/solarflow-bt-manager.py", line 232, in main
    asyncio.run(run(broker=mqtt_broker, port=mqtt_port, info_only=info_only, connect=connect, disconnect=disconnect, ssid=ssid))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/drschnalli/solarflow-bt-manager/src/solarflow-bt-manager.py", line 136, in run
    async with BleakClient(device) as bt_client:
  File "/usr/local/lib/python3.9/dist-packages/bleak/__init__.py", line 565, in __aenter__
    await self.connect()
  File "/usr/local/lib/python3.9/dist-packages/bleak/__init__.py", line 605, in connect
    return await self._backend.connect(**kwargs)
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/client.py", line 249, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Software caused connection abort`

Jemand eine Idee? Got any ideas?

reinhard-brandstaedter, 2024-02-21:
This seems like a BT connection lock-up.
You can try to reset the BT connection with ‘bluetoothctl’

Note: I do not recommend running bt-manager in bluetooth proxy mode for permanent production use. The connection with the hub is just too unstable…

tuxianerDE, 2024-04-11:
@drschnalli is this issue still present?

closed 2024-07-17 by reinhard-brandstaedter