reinhard-brandstaedter/solarflow-bt-manager #9: no telemetry data from hub in offline mode with Firmware Version 2.0.38

e-bits, 2024-03-25:
This weekend I did many Tests connecting my SmartHub1200 to a local MQTT Broker.
I found issue #8 which is very similar to my Issue but none of the Fixes helped me.

I'm able to connect my SmartHub to a local MQTT Broker and was even able to send a write Command to enable or disbale the BEEP.
```
Topic: 
iot/73bkTV/<device ID>/properties/write

Message:
{"timestamp": 1711317292706, "messageId": "08fd55741bf94752855fb4d35eff4794","deviceId": "<device ID>","properties":{"buzzerSwitch":1}}
```

I setup Wireshark to trace what is beeing sent during a connection to `mqtteu.zen-iot.com`

```
/73bkTV/<device ID>/connected
{"deviceKey":"<device ID>","onlineFlag":"00601"}
```

Did they probably change a confirmation messages after a successful join to enable sending telemetry Data which is currently missing in offline mode?

Currently I'm out of Ideas. Does anyone else have similar issues with **Firmware Version 2.0.38** in offline mode? Proxy Mode is working perfelcty fine.

How can I help to provide further details to get this issues analyzed better?
Many thanks for your support and this wunderful piece of Software.

Cheers e-bits

tuxianerDE, 2024-03-25:
Hey, are you talking about the SolarFlow Hub 1200 or is this again another Zendure product?
If you are able to connect to the MQTT and you are able to disable the beep then let me ask what is the issue? if you are able to switch off the beep, you are connect however the hub sends the telemetry data in a weird bulk json structure. There is mapping tool available that deconstructs the JSON into single line entries.

The message structure of the telemetry data you find /ThisIsBlank/73bkTV/deviceID/ 

Note the first path is BLANK

Are you running the mapping script? I can be found in [solarflow-topic-mapper.py](https://github.com/reinhard-brandstaedter/solarflow-bt-manager/blob/master/src/solarflow-topic-mapper.py)

e-bits, 2024-03-25:
No mapper script is currently not running because no data arrive to map. Yes it's the SolarFlow Hub 1200.
Attached you can find a Screenshot with the Data which is available in local MQTT Broker. For test reason only the SolarFlow Hub 1200 is configured to communicate with the local MQTT Broker no other devices.

[image]

If I understand everything correctly I should a least have some `properties` Topic's which can be mapped.

In the situation described in the Screenshot the device is able to process `write` Topics.

tuxianerDE, 2024-03-25:
Ok. So just to put your mind at ease, i am using firmware 2.0.38 as well and it is working for me. So we will figure it out.

Is your mosquitto running with authentication turned on or off? If it is turned on (or you are not sure) can oyu check the the authentication properly turned off, in the related issue you referenced this was the ultimate problem causes all the hassle? 
If it is turned off, (already) can you restart the mosquitto server as we have seen "dirty" connections to cause problems. 

Is your hub connected/like the wiring? I see a note in your message of "electric: 0" not sure what it is relating too, however I know from the other messages and my messages that the value was >0. Also I dont see an error message coming back from the triggered events, that looks so far not to bad.

e-bits, 2024-03-25:
> So just to put your mind at ease, i am using firmware 2.0.38 as well and it is working for me.

Great to hear, I really appreciate your help.

Im running my mosquitto with the following config in `mosquitto.conf`
```
listener 1883
allow_anonymous true
```
I think this means authentication is turned off right? Because the Mosquitto Broker for this test is specially for SolarFlow Hub 1200 it is turned off and on every time I try a reconnect.

I think it's connected correctly because it shows the values correctly in app as well in proxy mode.

tuxianerDE, 2024-03-25:
Yes and no, if the credentials line or password file line is present in you mosquitto it will accept a user and password and then deny accordingly. anonymous only works if nothing is populated at the time of request. Can you double check your config 

See https://github.com/reinhard-brandstaedter/solarflow-bt-manager/issues/8#issuecomment-2000416615

e-bits, 2024-03-25:
Hi @tuxianerDE,
this is everything I have in my mosquitto.conf, nothing else.
```
listener 1883
allow_anonymous true
```

e-bits, 2024-03-26:
I was reading trough this community post: https://community.symcon.de/t/zendure-smartflow-mqtt/134538/10

I'm using `solarflow-bt-manager` only to connect my SolarFlow Hub 1200 to a local MQTT Broker. All the logic for controlling I'm planning to do in Home-Assistant.

This is the command i'm using to disconnect SolarFlow Hub 1200 from Zendure Cloud and connecting locally: `python3 solarflow-bt-manager.py -d -w <WiFi SSID> -b <local MQTT broker>`

Could it be I need to manually send this MQTT Message to enable Telemetry reporting. Or do I need to pass the `-i` Option?

Topic: 
`iot/73bkTV/<device ID>/properties/read`

Message:
```json
{
    "properties": [
        "getAll"
    ]
}
```

I will try this, after work.

tuxianerDE, 2024-03-26:
I assume you execute the necessary exports before that too right? (just to be safe)

The command looks right! Did you make sure that no other Bluetooth device is trying to connect (i.e. Zendure App on the phone). 

How long did you wait for data to come in? The hub only sends data on sporadic intervals, hence the above message you posted is used to force it to talk to us. Yes you can execute this manually and trigger the output of information. If you execute the export and the skript as above everything looks right to get it connected. As said post the read message or you actually will need to wait a bit only the first data arrive.

e-bits, 2024-03-27:
Hi @tuxianerDE,

thanks for all your Support. Sending the Topic: `iot/73bkTV/<device ID>/properties/read` did the trick.
Because I recorded the credentials with Wireshark during connection setup to `mqtteu.zen-iot.com` I'm even able to connect with username and password to local MQTT. Which means no need for `allow_anonymous true` in `mosquitto.conf` in my case.

I will go on now and try automate things in Home-Assistant.

Cheers e-bits

closed 2024-03-27 by e-bits