reinhard-brandstaedter/solarflow-bt-manager #10: Workaround for Solarflow-BT-Manager

olomberg, 2024-03-26:
Hi Reinhard.

I just bought an implemented my Solarflow PV Hub 2000 and AB 2000 for my house. Zendure App is fully configured. Now I would like to use your Solarflow Control and Statuspage part. Unfortunately I am currently 3 hours away from my house but have remote access via vpn tunnel. Given that I am not able to reconfigure the PV Hub via bluetooth. I did a Wireshark sniff an can see the PV Hub sending it's MQTT packages to mqtteu.zen-iot.com as I did the initial setup an configuration to Germany. So while thinking about how I could re-configre it remotely, while not being onsite, I came up with the idea to just fake the dns entry of "mqtteu.zen-iot.com" on my local dns server (leveraged also by my PB Hub) pointing to my local Mosquitto mqqt server. Did somebody ever tried/tested this? Should this work with your control tool? I will test this tomorrow, when I will have more time. Feedback on "faking" DNS is appreceated.
Best
Oliver

olomberg, 2024-03-26:
Update: Seems to be working so far. Just had the chance to set a DNS record of "mqtt.zen-iot.com" pointing to my local mqtt server and disconnected my internet connection so that the PV Hub has to re-establish the connection. After this can see the mqtt messages flying in. At the same time my PV Hub is "Offline" in the Zendure App under My devices.

reinhard-brandstaedter, 2024-03-27:
Yes, that will work seamlessly.
This split-dns was also my initial setup without changes on the hub side. But since not everyone is able to operate their own DNS I took the generic approach within the hub’s config.

mavo, 2024-04-05:
Hello @olomberg,

as you have the hub2k for some days now, did you manage to have sf-control working with it? I am currently struggeling. Maybe you have an idea what I forgot to do?

https://github.com/reinhard-brandstaedter/solarflow-control/issues/191

ninjadriver, 2024-04-12:
> Yes, that will work seamlessly. This split-dns was also my initial setup without changes on the hub side. But since not everyone is able to operate their own DNS I took the generic approach within the hub’s config.

Hi,
could you push or send me an short information about how you started with faking the DNS Entry?
Background: 
- I have alread a TSUN Proxy in Place, which is operating on that way: DNS Entry is faked to my Server, Sends Information to MQTT and then Forward the information to the TSUN Server.
- Now i'm looking for the same solution without using BT. So I want to Fake DNS Entry für MQTT Connection to my local MQTT Server and i want to forward the incoming MQTT Messages to the MQTT Adress of Zendure. Do you know if this could work? Would be very happy if you have a solution for me :) Thanks in Advance

Mediacj, 2025-01-11:
Is this split-dns still possible? I recently tested it but couldn’t get it to work. Could this be related to authorization from Zendure?

Mediacj, 2025-01-11:
> Update: Seems to be working so far. Just had the chance to set a DNS record of "mqtt.zen-iot.com" pointing to my local mqtt server and disconnected my internet connection so that the PV Hub has to [...]

Is this still possible? I recently tested it but couldn’t get it to work. Could this be related to authorization from Zendure?

closed 2025-11-23 by reinhard-brandstaedter