reinhard-brandstaedter/solarflow-bt-manager #12: Solarflow-BT-Manager funktioniert nicht mit dem Zendure HUB-2000

Newbie-2, 2024-04-11:
Das Trennen = OFFLINE Mode mit den Solarflow-BT-Manager funktioniert nicht mit dem Zendure HUB-2000 !

In Zeile 40 ist der Befehl falsch:
 client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id="solarflow-bt")

Nur mit der Änderung in:
client = mqtt_client.Client(client_id="solarflow-bt")
kann der HUB-2000 angesprochen werden !

Desweiteren kommen über die Zendure Cloud per MQTT für den HUB-2000 die Informationen über den Pfad "/state"
Benutzerkonto/SF_DEVICE_ID/state
[image]

Leider kommen "nach" der Trennung im eigenen MQTT zB. MQTT-Explorer keine dieser Werte mehr an !?

[image]

**Wie können die Zendure Cloud-Werte unter /state, im MQTT-Explorer angezeigt werden und OHNE "solarflow-control" genutzt werden ?**

boennemann, 2024-04-11:
This is pretty much exactly my case, too. And @tuxianerDE I have to correct myself in #8 I said I'm on 3.0.8, while I meant 3.0.10.

Did 3.0.10 introduce this?

reinhard-brandstaedter, 2024-04-12:
> Das Trennen = OFFLINE Mode mit dem Solarflow-BT-Manager funktioniert **NICHT** mehr mit dem Zendure HUB-2000 !
> 
> In Zeile 40 ist der Befehl falsch: client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id="solarflow-bt")
> 
> Nur mit der Änderung in: **client = mqtt_client.Client(client_id="solarflow-bt")** kann der HUB-2000 angesprochen werden !

That is a dependency error that was introduced with the recent release of paho-mqtt version 2.0
The first one works with 2.0 whereas the latter one is for version 1.6.x. You have to ensure the correct version is installed/used on your setup.

Newbie-2, 2024-04-12:
Hello Reinhard,
Thanks for your answer.

Unfortunately I don't know how to fix the error???

1.)   I have installed the current BT manager
git clone https://github.com/reinhard-brandstaedter/solarflow-bt-manager.git

2.)   I changed line 40 in the file: solarflow-bt-manager.py as follows: client = mqtt_client.Client(client_id="solarflow-bt")

3.)   I have installed Python version 2.00
sudo pip3 install paho-mqtt==2.0.0

The result of the separation looks good to me:

[image]

4.)   Press the IO button for 6 seconds (no LED lights up)
       Press the IO button for 1 second (left and right LEDs light up)

But there are still no values in the MQTT Explorer!

[image]

Which steps in the process need to be changed so that I can receive the data from the HUB-2000 in the MQTT Explorer, as with the HUB-1200?

Newbie-2, 2024-04-12:
[image]

After successfully disconnecting, the HUB-2000 only had to be shut down completely (press the IOT button for 10 seconds).
Then disconnect the battery and solar panels for 10 seconds. After a restart, the data was displayed under “report”.

Newbie-2, 2024-04-12:
After successfully disconnecting, the HUB-2000 only had to be shut down completely (press the IOT button for 10 seconds).
Then disconnect the battery and solar panels for 10 seconds. After a restart, the data was displayed under “report”.

closed 2024-04-12 by Newbie-2

boennemann, 2024-04-13:
@Newbie-2 Thanks for this recommendation. I didn't occur to me to also disconnect the battery and modules 🤦 

On the plus side: I'm now receiving a ton of calls from the hub 👍 
On the down side: It's just a bunch of errors and no data. 

The Hub is now publishing an error message to MQTT every second `/A8yh63/<ID>/event/error`

Sometimes it includes an error code, too. 30, 217, 218 is what I've seen so far.

boennemann, 2024-04-13:
OMG, in the moment I posted this all of a sudden it started rolling in. Thank youuu 🥹🥹🥹