reinhard-brandstaedter/solarflow-bt-manager #13: Verbindungs und "payload" Probleme

martin-nrw-85, 2024-04-12:
Hallo,
ich hoffe hier kann man mir helfen.
Ich habe den Hub1200 offline genommen. 
Musste eine Änderung in dem Skript vornehmen:
- original:
    client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id="solarflow-bt")

- Anpassung:
    client_id= "solarflow-bt"
    client = mqtt_client.Client(client_id)

Da es immer zu einer Fehlermeldung:
"  File "/zendure/solarflow-bt-manager/src/solarflow-bt-manager.py", line 43, in                               local_mqtt_connect
    client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_                              id)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'paho.mqtt.client' has no attribute 'CallbackAPIVersion'"

Im MQTT Explorer bekomme ich jedoch nur "0{"messageId":"123","method":"device","deviceId":"xxxxx","timestamp":1712852073,"offData":0,"data":{"eventId":3,"startTime":1712851893,"endTime":1712852073,"inout":2,"mode":2,"electric":0}}"
angezeigt sonst kommen keine Ereignisse.

_Originally posted by @martin-nrw-85 in https://github.com/reinhard-brandstaedter/solarflow-bt-manager/issues/10#issuecomment-2050250540_

reinhard-brandstaedter, 2024-04-12:
Starte den hub mal durch damit er sich neu anmeldet.
Wenn beim offline nehmen alles ohne fahler quittiert wurde kommt er im normalfall.

martin-nrw-85, 2024-04-12:
Hi danke werde ich übers Wochenende mal probieren.

ricariel, 2024-04-12:
> Da es immer zu einer Fehlermeldung:
> " File "/zendure/solarflow-bt-manager/src/solarflow-bt-manager.py", line 43, in local_mqtt_connect
> client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_ id)

Das liegt an der paho-mqtt library, welche ich benutzt habe. War Version 2. Hätte aber die Version 1.* sein müssen um kompatibel zu sein mit der Vorversion.

ricariel, 2024-04-12:
Ja https://github.com/reinhard-brandstaedter/solarflow-control tut das zum Beispiel.

PR #14 (2024-04-13): Fix timing problem on disconnect
Fixes #13
closed 2024-04-13 by reinhard-brandstaedter