reinhard-brandstaedter/solarflow-bt-manager #18: Step-by-Step guide for Pi Zero

lemorak, 2024-05-15:
Hi,

I'm having trouble setting up the script on a Pi Zero. Is it possible to get a step-by-step guide on how to install the script? I've already installed raspbian an copied the files over but now I don't know what do to. If I try to start the script I get the following error:

python3 solarflow-bt-manager.py -i

Traceback (most recent call last):
  File "/home/sir_lemorak/Downloads/solarflow-bt-manager-master/src/solarflow-bt
-manager.py", line 4, in <module>
    from bleak import BleakClient, BleakScanner
ModuleNotFoundError: No module named 'bleak'

Can you please help me here? And when everything is working, how do I set up an autostart for the script after rebooting the Pi?

Thanks and greetings
lemorak

reinhard-brandstaedter, 2024-05-15:
[These instructions](https://github.com/reinhard-brandstaedter/solarflow-bt-manager?tab=readme-ov-file#disconnecting-the-hub-from-the-cloud) should be really all that is needed on a raspberry OS that has python already installed.

There is no need to keep the script running after a reboot, typically you want to disconnect your hub once and that's it. I do not recommend to keep it running in bluetooth mode for longer.

lemorak, 2024-05-15:
Thanks for the fast reply. The problem is that I'm stuck at the very first step "getting basic information". Here I get the error form the post above. What am I doing wrong? As I said I copied over the files and try to run the script.

Why don't you recommend running bluetooth mode for longer? I want to have my data local and not in a china cloud ^^

Greeting
lemorak

lemorak, 2024-05-15:
Update:
I got it working but my problem now is that I use Home Assistant with Mosquitto and an anonymous login is not possible.

lemorak, 2024-05-15:
Thanks for the help! Now via bluetooth it works but only as long as I have my putty terminal window open. As soon as I close it the mqtt server doesn't recieve data any more.

Connection over wifi doesn't work. You said "Unfortunately it will use the hubs default (hardcoded?) password." Is this the problem here because my mosquitto server doesn't allow anonymous connections? Do the variables "$ export MQTT_USER="your mqtt user" and $ export MQTT_PWD="mqtt password" only work for bluetooth?

reinhard-brandstaedter, 2024-05-21:
@lemorak running in bluetooth mode/proxy mode connects to the hub and then forwards telemetry to MQTT, this is why you can specify a user/pwd there.
But the "production mode" (fully disconnecting the hub) means the hub will directly report to MQTT. If you can't allow anonymous access to your broker you can send me your device ID (your should find it in the output of the bt-manager) I can then tell you which user you need to add to your MQTT broker so that the hub can connect.

To your first question: closing the ssh connection of course terminates every process running in foreground, but as said before: the bluetooth mode is not intended for a permanent operation anyways...

lemorak, 2024-05-22:
@reinhard-brandstaedter I'm using Home Assistant with the miquitto addon where the anonymous access can't be enabled. 
My device ID is: W8R7207C

At the moment I'm using tmux to let the process running but as you said it's not a permanent solution. I have to restart the Pi after some days.

Thanks again for your help!

reinhard-brandstaedter, 2024-05-22:
@lemorak any way I can PN you your user/pwd?

lemorak, 2024-05-22:
@reinhard-brandstaedter you can PN me on X/Twitter. @Sir_Lemorak

closed 2024-05-27 by lemorak

rosch100, 2024-07-07:
@reinhard-brandstaedter ich habe das gleiche Problem. Wäre toll, wenn du mir für die Device-ID: esAG445A auch die Zugangsdaten für mqtt schicken könntest.
X: @Rol6674
Oder auch irgendwo in München gegen ein &#127866;

Sascha984, 2024-08-02:
@reinhard-brandstaedter  Ich habe das gleiche Problem. Wäre toll, wenn du mir für die Device-ID: s3W4a8sm auch die Zugangsdaten für MQTT schicken könntest.