reinhard-brandstaedter/solarflow-bt-manager #19: No Solarflow device found

joe63, 2024-06-08:
I try to take my SF-Hub offline but I don't get a connection.

I tried all four items (move closer, restart solarflow, disconnect mobile app, restarted the raspi several times).

But I don'z now the meaning of [**DEL**] Device XX:C9:60:AD:F1:6F ZenH1_11  ??

Points it to the Problem?

Here I have the screen of the command to start the bt-manager and the answer from bluetoothctl.

pi@raspberrypi:~ $ python3 solarflow-bt-manager.py
2024-06-08 11:01:36,266:INFO: scan for: zenp
2024-06-08 11:01:47,666:INFO: No Solarflow device found! You can try these steps:
                   - Move closer to the hub
                   - Reset your bluetooth connection (bluetoothctl)
                   - Restart the Solarflow Hub
                   - Disconnect any mobile Apps currently connected to the hub
                   - 
pi@raspberrypi:~ $ bluetoothctl
Agent registered
[CHG] Controller B8:27:EB:EB:FE:9A Pairable: yes
[DEL] Device XX:D0:B4:31:07:15 XX-D0-B4-31-07-15
[DEL] Device XX:26:9E:17:7F:EA XX-26-9E-17-7F-EA
[DEL] Device XX:E5:0B:16:9C:39 XX-E5-0B-16-9C-39
[DEL] Device XX:D1:C3:02:B7:15 XX-D1-C3-02-B7-15
[DEL] Device XX:EC:61:A9:0B:69 Smart Tag
[DEL] Device XX:16:DE:C9:28:A0 XX-16-DE-C9-28-A0
[DEL] Device XX:1D:4C:52:92:A9 XX-1D-4C-52-92-A9
[DEL] Device XX:F1:57:4C:2B:38 XX-F1-57-4C-2B-38
[DEL] Device XX:AD:9B:18:AF:99 XX-AD-9B-18-AF-99
**[DEL] Device XX:C9:60:AD:F1:6F ZenH1_11**

What could be the problem?

reinhard-brandstaedter, 2024-06-08:
Seems you are using a Hub2000. Try setting the environment variable SF_PRODUCT_ID="A8yh63"
Then it will scan for a different device string:

```2024-06-08 11:01:36,266:INFO: scan for: zenh```

joe63, 2024-06-08:
Thank you, it works now!

_pi@raspberrypi:~ $ python3 solarflow-bt-manager.py
2024-06-09 00:49:10,016:INFO: scan for: zenh
2024-06-09 00:49:11,465:INFO: Found device: 94:C9:60:BB:F1:6F: ZenH1_11_

closed 2024-06-09 by reinhard-brandstaedter