reinhard-brandstaedter/solarflow-bt-manager #20: can no more connect to cloud

joe63, 2024-06-10:
**I tried to disconnect my HUB2000.**

in the mqtt-explorer I only saw this:
iot
    A8yh63
        W7798JGG
            register
                **replay = {"messageId":123,"timestamp":1718034560,"params":{"token":"abcdefgh","result":0}}**

nothing else!

so I wanted to **reconnect. it doesn't work.**
also with shuting down the HUB2000 with waiting 10s and cut panels and batterie, ....

as i tried once more the command:  **python3 solarflow-bt-manager.py -i**

---> I got error which I didn't see as I run this command before trying to disconnect.

**python3 solarflow-bt-manager.py -i**
_2024-06-10 19:36:56,077:INFO: scan for: zenh
2024-06-10 19:36:57,152:INFO: Found device: 94:C9:60:AD:F1:6F: ZenH1_11
2024-06-10 19:36:57,978:INFO: Services:
2024-06-10 19:36:57,992:INFO: 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
2024-06-10 19:36:58,003:INFO: 00112233-4455-6677-8899-aabbccddeeff (Handle: 18): Unknown
2024-06-10 19:36:58,007:INFO: 0000a002-0000-1000-8000-00805f9b34fb (Handle: 12): Vendor specific
2024-06-10 19:36:58,096:INFO: {'method': 'report', 'deviceId': 'Wxxxxxxx', 'properties': {'wifiState': 1, 'solarInputPower': 52, 'outputHomePower': 48}}
2024-06-10 19:36:58,258:INFO: {'method': 'report', 'deviceId': 'Wxxxxxxx', 'properties': {'solarPower2': 18, 'solarPower1': 34}}
2024-06-10 19:36:58,518:INFO: {'messageId': '123', 'method': 'getInfo-rsp', 'deviceId': 'Wxxxxxxx', 'timestamp': 1718041014, 'deviceSn': 'H**************', 'firmwares': [{'type': 'MASTER', 'version': 12298}, {'type': 'BMS', 'version': -1}, {'type': 'BMS_AB2000', 'version': 4110}]}
2024-06-10 19:36:58,523:INFO: The SF device ID is: Wxxxxxxx
2024-06-10 19:36:58,524:INFO: The SF device SN is: H**************
2024-06-10 19:36:58,659:INFO: {'method': 'report', 'deviceId': 'Wxxxxxxx', 'properties': {'solarInputPower': 48, 'outputHomePower': 44, 'solarPower2': 17}}
2024-06-10 19:36:58,776:INFO: {'messageId': '123', 'method': 'error', 'deviceId': 'Wxxxxxxx', 'timestamp': 1718041015, 'offData': 1, 'data': []}
2024-06-10 19:36:58,922:INFO: {'method': 'report', 'deviceId': 'Wxxxxxxx', 'properties': {'solarPower1': 31}}
2024-06-10 19:36:59,761:INFO: {'method': 'report', 'deviceId': 'Wxxxxxxx', 'properties': {'solarInputPower': 42, 'outputHomePower': 38, 'solarPower2': 15}}
2024-06-10 19:37:00,061:ERROR: A message handler raised an exception: Extra data: line 1 column 74 (char 73).
Traceback (most recent call last):
  File "src/dbus_fast/message_bus.py", line 811, in dbus_fast.message_bus.BaseMessageBus._process_message
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/manager.py", line 874, in _parse_msg
    on_value_changed(message.path, self_interface["Value"])
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/client.py", line 172, in on_value_changed
    callback(bytearray(value))
  File "/home/pi/solarflow-bt-manager.py", line 101, in handle_rx
    payload = json.loads(data.decode("utf8"))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 340, in decode
    raise JSONDecodeError("Extra data", s, end)
json.decoder.JSONDecodeError: Extra data: line 1 column 74 (char 73)

2024-06-10 19:37:00,179:INFO: {'messageId': '123', 'method': 'getInfo-rsp', 'deviceId': 'Wxxxxxxx', 'timestamp': 1718041016, 'deviceSn': 'H**************', 'firmwares': [{'type': 'MASTER', 'version': 12298}, {'type': 'BMS', 'version': -1}, {'type': 'BMS_AB2000', 'version': 4110}]}
2024-06-10 19:37:00,181:INFO: The SF device ID is: Wxxxxxxx
2024-06-10 19:37:00,182:INFO: The SF device SN is: H**************
2024-06-10 19:37:00,761:INFO: {'method': 'report', 'deviceId': 'Wxxxxxxx', 'properties': {'solarInputPower': 38, 'outputHomePower': 34, 'solarPower2': 13}}_

did I kill the HUB2000?

any idea??

reinhard-brandstaedter, 2024-06-10:
I don't think you killed your hub. The output of the -i command looks good enough, I wouldn't worry too much about the decode error.
So your hub sends data fine via bluetooth, you can always re-initialize it with the Zendure app.

But before you do that, are you sure there is no topic with a empty first node in your MQTT (first line bwlow)?
[image]

Sometimes it takes quite a bit until the hub sends it's first telemetry, sometimes a restart is required as well.
When you shut down your hub, did you press the button 10s or 6s?
Did you try to connect via BT in the Zendure App (note that bluetooth is only available after startup and no connection to WiFi for some time)

joe63, 2024-06-11:
Hi,

no there a not more mqtt-info, even after 5 minutes
[image]

i pressed the button min 10sek.

**is this an error or a warning?**
pi@raspberrypi:~ $ python3 solarflow-bt-manager.py -d -w l******* -b 192.168.1.144
Disconnecting Solarflow Hub from Zendure Cloud
2024-06-11 08:37:35,042:INFO: scan for: zenh
2024-06-11 08:37:36,363:INFO: Found device: 94:C9:60:AD:F1:6F: ZenH1_11
2024-06-11 08:37:37,709:INFO: Services:
2024-06-11 08:37:37,712:INFO: 00112233-4455-6677-8899-aabbccddeeff (Handle: 18): Unknown
2024-06-11 08:37:37,715:INFO: 0000a002-0000-1000-8000-00805f9b34fb (Handle: 12): Vendor specific
2024-06-11 08:37:37,718:INFO: 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
**/home/pi/solarflow-bt-manager.py:41: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id="solarflow-bt")**
2024-06-11 08:37:37,787:INFO: {"iotUrl": "192.168.1.144", "messageId": "1002", "method": "token", "password": "*************************", "ssid": "l******", "timeZone": "GMT+02:00", "token": "abcdefgh"}
2024-06-11 08:37:37,809:INFO: Setting IoTURL connection parameters - disconnect

reinhard-brandstaedter, 2024-06-11:
That is just a warning...
Are you sure your hub is connected to WiFi and can reach your MQTT broker. Did you check your MQTT server's logs? For connection attempts?

joe63, 2024-06-11:
I'm even not sure if the hub is connected to WiFi. 
How can I check this? In the DHCP Clients List from my router? 
There I don't see a special client name.

reinhard-brandstaedter, 2024-06-11:
Yes in your DHCP client list you should definitely see your hub.
Did you specify your WiFi password correctly via [environment variable ](https://github.com/reinhard-brandstaedter/solarflow-bt-manager?tab=readme-ov-file#disconnecting-the-hub-from-the-cloud) when configuring/disconnecting the hub?

What you also can try is join the hub to another WiFi...eg. mobile hotspot before joining it to your standard WiFi.

joe63, 2024-06-11:
Hi, in the meantime I found the hub in my router (with the mac-address).

[image]

this shows wireshark:
[image]

[image]

[image]

reinhard-brandstaedter, 2024-06-11:
I could be wrong but this looks like your MQTT broker doesn't allow the hub to subscribe.
Are you sure you created the correct user on the MQTT side or allowed anonymous access? (i'd go with anonymous access to rule out other issues)

joe63, 2024-06-12:
Hallo, you were absolutely right. Danke!
I now allowed anonymous access and kann see the data in mqtt-explorer.

what I have to change to see the data in the sf-statuspage?

joe63, 2024-06-12:
Hallo, you were absolutely right. Danke!
I now allowed anonymous access and kann see the data in mqtt-explorer.
under   /A8yh63/W*********/properties/report 

what I have to change to see the data in the sf-statuspage? 
start it without the parameter --online , didn't bring success

reinhard-brandstaedter, 2024-06-12:
That is described [here](https://github.com/reinhard-brandstaedter/solarflow-statuspage/blob/master/README.md)

joe63, 2024-06-12:
Hello,
sorry I forgot to say that it was already running in online-mode and showing values.
but now on offline-mode there are no values.

reinhard-brandstaedter, 2024-06-12:
Ho did you start it? Environment variables set?

joe63, 2024-06-12:
[image]
 in case of mqttuser/pw I tried several values, user and password which I use normaly when I access my iobrocker and also Product_id and pw which I got from you

joe63, 2024-06-12:
with a nodered -  workaround  I can show values in the status-page.
for example the solar-input
[image]

[image]

function
_let value = msg.payload["properties"];
let newmsg = {};
newmsg.payload = {};

if (value["outputHomePower"] != null) {
   newmsg.payload = value["outputHomePower"];
}
else return null
return newmsg;_

but I think this is not the sense of your tool.

joe63, 2024-06-12:
one strange thing was also that in online-mode I saw the SoC in the openDTU-tool.
in offline mode when I changed the mqtt-topic to /A8yh63/Wxxxxxxx/properties/report/electricLevel I saw no value.
but when I published a value (e.g.87 ) with the mqtt-explorer I saw it in the openDTU section.

at the moment I receive the mqtt-value in nodered and publish it  once more with nodered and it is visble in openDTU.

[image]

is it an access right-issue?

joe63, 2024-06-12:
I have the intention to run solarflow-control. I wanted to do it step by step. 
**Thank you!!**

one last question for today: what is the meaning from red lines in wireshark? (error, warning,...)
background is : my screenshot from yesterday (upscrolling some screens) shows a detail from the try to reconnect the hub ( which does not work). And there are red TCP-protocol-lines. I have to investigate this better next weekend.

joe63, 2024-06-13:
when running  sf-control I get red, orange and green lines!  --> I see no value in the Home Power Meter section!

[image]

to the green smartmeter-line:
looks so in the mqtt-explorer: (I asked you some weeks ago)
[image]

and so in the config.ini:
[image]

can you pls help me (again)?

closed 2024-06-18 by reinhard-brandstaedter