reinhard-brandstaedter/solarflow-bt-manager #21: mqtt Mosquitto

sturowski, 2024-06-19:
Hallo, 
erstmal sehr cooles Projekt und ich freue mich sehr drauf, das zu benutzen sobald meine Anlage steht. 
Gibt es was neues bezüglich dem Mqtt Username und Password vom hub? Kann man das mittlerweile setzen? Wenn nicht, muss ich meinen mqtt komplett offen machen oder gibt es die Möglichkeit nur wegen Username ohne PW einzutragen für den hub? Geht sowas? 
Hast du schonmal drüber nachgedacht einen zweiten mqtt Servern (ohne Authentifizierung) zu betreiben und mit einem kleinen Python Script dann Daten vom Transport mqtt in den normales (mit Authentifizierung) zu sorgen m schieben? Oder ist das ganz egal, weil der mqtt eh im lokalen Netz läuft und das sicher genug ist?

reinhard-brandstaedter, 2024-06-25:
Hi @sturowski ,
you can calculate the MQTT password from the device ID. With that you can also add the user/pwd to your existing mosquitto server. I won't publish the way how this password is calculated.
Another way would be to use a network sniffer to decipher mqtt traffic and get the password from there.

closed 2024-06-25 by sturowski

bribo129, 2024-07-26:
Hello @reinhard-brandstaedter,
I also use HA with Mosquitto. Unfortunately, I haven't managed to calculate or sniff the password yet. Could you help and send me the password?

device ID is: AkA499J5
contact: bb.temp@outlook.de

psw005, 2024-10-27:
Hello @reinhard-brandstaedter,
same here - I use HA with MQTT and authentication, too. 
I managed to get wireshark running and logged the traffic from the local Solarflow-IP towards Zendure. But I couldn't find  the user/ password yet. Could you please give me a hint or calculate the password?

device ID is: Qg41VSfL
contact: psw5@gmx.de

Thanks a lot!

psw005, 2024-11-13:
Hello @reinhard-brandstaedter,
Could you please help me regarding the mqtt-user/pw? I tried several options from what I discovered by using wireshark without success so far. 

Thank you!