reinhard-brandstaedter/solarflow-bt-manager #22: No /report topics from Hub2000 in offline mode

nfocke, 2024-07-08:
I have been using my Hub1200 / "old" hub in offline mode = my own MQTT broker for some time. This works well, also taking it back to the Zendure MQTT and to any MQTT broker works fine.

I recently acquired a Hub2000 and now want to use it in the same way. I can, as for the Hub1200, direct it to another MQTT broker using the  {"iotUrl": ...   } / {"method": "station"} GATT message as for the Hub1200. It does register with the broker, and publishes registration and event topics. It also responds to the "getAll" messages as usual. However, it does not send and /report topics. It seems Zendure has made some changes to the connection logic.

Has someone succeeded in bringing a Hub2000 "offline"? And if so, did it work for you directly as implemented in the solarflow-bt-manager or did you have to change the registration procedure? Rebooting the hub did not work either. I also made sure, all Bluetooth connections are closed / BT set to offline.

reinhard-brandstaedter, 2024-07-08:
Check the discussions at solarflow-control, there are quite some hub2k users with offline hubs. In that regard they aren’t different.
What firmware are you on?

nfocke, 2024-07-08:
I am on the most recent V3.0.13 (for the Hub2000). When checking the discussion on the solarflow-control page, I am not so sure if there are many successful user. I only found 3 topics, and only one user [joe63](https://github.com/joe63) with specific questions. 
I just re-checked with the current release from the Github and get the same behavior.  I can pull the hub to register with the MQTT broker and get some topics:
[image]
But no further info is coming in.

Interestingly, the re-connecting to Zendure cloud also seems to have the same issue. The hub is shown online, but no data updates are coming in and no control is possible. So it seem that something else needs to be going on during "official" hub registration, at least for hub2000 and the V3.0.13 firmware. Will investigate further.

ricariel, 2024-07-09:
Works Fine Herr with Hub 2000 and 3.0.13. did. you reboot the Hub?

nfocke, 2024-07-09:
@ricariel Good to hear it works. Yes, I did reboot the hub (6 seconds IoT, switch on again). But no difference. Did you do the switch with the 3.0.13 firmware and the new dark mode app? Or did you do it previously and updated the hub later?

ricariel, 2024-07-09:
I did it both. I had to do it twice since last Firmware update but it works after reboot

nfocke, 2024-07-09:
@ricariel interesting. I have rebooted twice (but after different back and forth MQTT broker switches) and have waited for some minutes.
So it did start reporting directly after the reboot (or did it take time? do you remeber?) and also did not report the property topics before the reboot?

reinhard-brandstaedter, 2024-07-09:
@nfocke a few things to check:
Is your MQTT allowing connections for anonymous users? If not, try to allow these.
Can you maybe do a network capture to track any connection attempts from the Hub?

closed 2024-07-09 by nfocke