reinhard-brandstaedter/solarflow-bt-manager #23: Solarflow Hyper2000

sweti3, 2024-07-09:
Hei,

firstly thanks for this amazing repo! Awesome what one can do! I have a question - do you know - or anyone - if it is compatible with the new SolarFlow Hyper 2000? I think it would use the same protocols (Why develop it twice??) - but I am not sure... Perhaps someone could test this? I plan on buying one, but only if I could control it myself.....

thanks a lot in advance!
Sven

reinhard-brandstaedter, 2024-07-09:
Hi Sven,
you also might want to follow [this thread](https://github.com/reinhard-brandstaedter/solarflow-control/issues/272)

Raudi1, 2024-07-20:
I have a Hyper2000 and a Pi zero 2 W near it. I also have MQTT access through the Zendure cloud. I'd be willing to help, but while I have good general Computer knowledge including Linux and can program in Python, I'm no developer.

Raudi1, 2024-07-20:
Got it to connect. Added the PRODUCT_ID to the code and pressed the button for 3 seconds. Afterwards I could connect over Bluetooth.

doppelgrau, 2024-08-09:
Currently thinking if I want to buy a Hyper 2000 or a Hub 2000 - someone experience with the local mode with the hyper 2000?
Would really like to be able control it via HomeAssistant/MQTT and even if I don't need currently the bidirectional charging, it would be a nice to have and also the integrated setup would look a bit "neater" than the hub with an additional inverter.

bbindreiter, 2024-08-13:
ACE 1500 also works fine once the product id is known.

linuxstony, 2024-08-21:
@Raudi1 which PRODUCT_ID was this? I would love to give this a try because the MQTT connection to the zendure one is slow at best and unreliable. Were you able to disconnect the Hyper completely and get the data via your local MQTT broker?

Raudi1, 2024-08-21:
> @Raudi1 which PRODUCT_ID was this? I would love to give this a try because the MQTT connection to the zendure one is slow at best and unreliable. Were you able to disconnect the Hyper completely and get the data via your local MQTT broker?

I made a pull request and it was merged. You can also see it in the official MQTT structure.
I was on holiday and had no time afterwards to do it yet. Perhaps I will have time to do it this weekend.

tuxianerDE, 2024-10-09:
hey @Raudi1 @bbindreiter  could you please also update the documentation page where the product IDs of the SF and AIO are documented so far? 

A to provide better transparency and then we can also close the issue. Thanks for hte support and work that more products can get offline

closed 2025-02-26 by reinhard-brandstaedter