reinhard-brandstaedter/solarflow-bt-manager #25: Possible bug in solarflow-topic-manager

DietmarSi, 2024-07-29:
Hi Reinhard,

First of all, thank you for the tools for Zendure products.

After I successfully switched my SolarFlow Hub 1200 to my local MQTT broker, I wanted to use the solarflow-topic-manager. I encountered an error, at least it was an error in my system:

**Line 143 (prevously):**
`report_topic = f'/{sf_product_id}/+/properties/report'`

**Line 143 (New):**
`report_topic = f'/{sf_product_id}/{sf_device_id}/properties/report'`

After this fix the solarflow-topic-manager works fine.

Greetz,
Dietmar

reinhard-brandstaedter, 2024-08-06:
Hi Dietmar,

the topic manager is kind of obsolete. Its purpose is now already included in sf-control alltogether.

kr,
Reinhard

ricariel, 2024-08-06:
If wished i can maintain the topic manager. I use it because it does everything i need.
Are patches welcome or do you plan to remove it?

Regards
Fabrice

DietmarSi, 2024-08-07:
Hi Reinhard,

from point of view that solarflow-control has included that is okay. But I do not use solarflow-control yet, because I have my own automation in Home Assistant to control the inverter.

What could be fine if solarflow-control will be added to the Addon Store for Home Assistant or HACS.

Kind regarde,
Dietmar

reinhard-brandstaedter, 2024-08-07:
@DietmarSi what exactly is the error you are getting?

the '+' in the report topic subscription should work as a placeholder just fine, working for any device_id.

DietmarSi, 2024-08-11:
@reinhard-brandstaedter  please excuse my late reply. I have lost the log messages  😞. 
If the + is only a placeholder, then I don't understand the issue.