reinhard-brandstaedter/solarflow-bt-manager #30: ModuleNotFoundError: No module named 'bleak' - when starting python3 solarflow-bt-manager.py -i

ralle6677, 2024-12-10:
Hi everyone,

 I’m currently trying to reconfigure my Hub 1200. When I run the script solarflow-bt-manager.py -i, I get the message 

Traceback (most recent call last):
  File "/Users/ralf/Documents/solarflow-bt-manager.py", line 4, in <module>
    from bleak import BleakClient, BleakScanner
ModuleNotFoundError: No module named 'bleak'

 I’ve tried running the script on both a Raspberry Pi and a MacBook, with the same result. Do you have any tips for me?

Danke und VG!
Ralf

ricariel, 2024-12-10:
Run this from the README

pip3 install -r requirements.txt

ralle6677, 2024-12-29:
Hello again,

when I disconnect the HUB I get these Message:

Disconnecting Solarflow Hub from Zendure Cloud
2024-12-29 15:54:13,689:INFO: scan for: zenp
2024-12-29 15:54:14,218:INFO: Found device: 99DA838C-0F2A-60F9-6BDE-A2B441542505: ZenP1_01
2024-12-29 15:54:15,935:INFO: Services:
2024-12-29 15:54:15,935:INFO: 0000a002-0000-1000-8000-00805f9b34fb (Handle: 12): Vendor specific
2024-12-29 15:54:15,935:INFO: 00112233-4455-6677-8899-aabbccddeeff (Handle: 18): Unknown
/Users/ralfmoosbrugger/Downloads/solarflow-bt-manager.py:41: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id="solarflow-bt")
2024-12-29 15:54:15,959:INFO: {"iotUrl": "192.168.178.177", "messageId": "1002", "method": "token", "password": "xxxxxx", "ssid": "xxxxxx", "timeZone": "GMT+02:00", "token": "abcdefgh"}
2024-12-29 15:54:15,960:INFO: Setting IoTURL connection parameters - disconnect

Is this normal? Cause I get no Topics in the EMQX-Broker.

Thank you for your help!

reinhard-brandstaedter, 2025-01-08:
Does your broker need authentication?