reinhard-brandstaedter/solarflow-bt-manager #34: Zendure SuperBaseV

Blubb10287, 2025-01-20:
Hi, I have a question: 
Is it possible to implement a Zendur SuperBaseV for the Mqtt as well, 
if so is it the same as described above or do I need different values?
If so, do you have any idea how to do it?

Unfortunately I am not very good at this.

Thank you very much.

reinhard-brandstaedter, 2025-01-21:
Likely a similar approach will work, with a different product ID.
I don't have a SuperBaseV so I can't investigate/experiement with it.

gh-fx2, 2025-03-17:
superbaseV 4600 :  productID=4y9q7a5E

gh-fx2, 2025-03-18:
@reinhard-brandstaedter:
Hi,
I need some help making superbaseV cloud-free.
a) 
export SF_DEVICE_ID=b74orp06
export SF_PRODUCT_ID=4y9q7a5E
export MQTT_USER=muser
export MQTT_PWD=mpwd
export WIFI_PWD=wlanpwd
...
python3 solarflow-bt-manager.py -c -w Wlan1
Connecting Solarflow Hub Back to Zendure Cloud
2025-03-18 20:02:28,578:INFO: scan for: zen
2025-03-18 20:02:28,661:INFO: Found device: DC:54:75:5E:2B:4E: ZenV4_26
2025-03-18 20:02:29,426:INFO: Services:
2025-03-18 20:02:29,426:INFO: 0000a002-0000-1000-8000-00805f9b34fb (Handle: 40): Vendor specific
2025-03-18 20:02:29,426:INFO: 0000a003-0000-1000-8000-00805f9b34fb (Handle: 64): Vendor specific
2025-03-18 20:02:29,426:INFO: 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
2025-03-18 20:02:29,427:INFO: {"iotUrl": "mq.zen-iot.com", "messageId": "1002", "method": "token", "password": "wlanpwd", "ssid": "Wlan1", "timeZone": "GMT+02:00", "token": "abcdefgh"}
2025-03-18 20:02:29,432:INFO: Setting IoTURL connection parameters - connect
prompt $
# it looks like working
b) 
python3 solarflow-bt-manager.py -d -w Sprite -b 192.168.28.28
Disconnecting Solarflow Hub from Zendure Cloud
2025-03-18 20:08:48,269:INFO: scan for: zen
2025-03-18 20:08:48,326:INFO: Found device: DC:54:75:5E:2B:4E: ZenV4_26
2025-03-18 20:08:51,078:INFO: Services:
2025-03-18 20:08:51,078:INFO: 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
2025-03-18 20:08:51,078:INFO: 0000a003-0000-1000-8000-00805f9b34fb (Handle: 64): Vendor specific
2025-03-18 20:08:51,078:INFO: 0000a002-0000-1000-8000-00805f9b34fb (Handle: 40): Vendor specific
Traceback (most recent call last):
  File "/home/ml/solarflow-bt-manager/src/solarflow-bt-manager.py", line 260, in <module>
    main(sys.argv[1:])
  File "/home/ml/solarflow-bt-manager/src/solarflow-bt-manager.py", line 257, in main
    asyncio.run(run(broker=mqtt_broker, port=mqtt_port, info_only=info_only, connect=connect, disconnect=disconnect, ssid=ssid))
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/ml/solarflow-bt-manager/src/solarflow-bt-manager.py", line 165, in run
    mq_client = local_mqtt_connect(broker,port)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ml/solarflow-bt-manager/src/solarflow-bt-manager.py", line 41, in local_mqtt_connect
    client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id="solarflow-bt")
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'paho.mqtt.client' has no attribute 'CallbackAPIVersion'
# this will not work

Question: any idea how I can continue ?
( PS: no python knowledge )

gh-fx2, 2025-03-19:
# solved:
my installed python3-paho-mqtt (V1.6.1) can not handle CallbackAPIVersion.
Simple change line 41 to
client = mqtt_client.Client(client_id="solarflow-bt")
has solved the problem.
my superbaseV is now cloud-free :)   thanks