reinhard-brandstaedter/solarflow-bt-manager #36: (Dis)Connection problem

I-H-deBoer, 2025-02-01:
Dear all, 

First my settings:
- HUB1200
- OpenDTU
- mosquitto with user/pw in Docker on Synology NAS
- MQTT Explorer on Windows 11
- Python3.12.8, CDM-line on Windows 11

I run the scrip with necessary settings to get offline with my WLAN & MQTT settings in CMD line, eg. "SET WIFI_PWD=<...>" and so on.

[image]

The MQTT Explorer looks like this:

[image]

I would expect that I would see more messages in MQTT explorer.  Solarflow-statuspage is - of course - also not working.
When I want o bring it back online to Zendure Cloud, WLAN cannot be restored, ZendureApp can only connect via BT to hub.
And, after I am online again, the MQTT explorer still looks the same. The "iot" entry does not disappear.

Any hints to solve this? Or did I miss something in the setup/doing?

Thanks.

reinhard-brandstaedter, 2025-02-01:
Did you create a user/pwd for the hub user on your mosquitto?
The hub uses a hardcoded username and password to authenticate, which cannot be changed.

If you cannot enable anonymous login to your MQTT, you will have to create a user with the hardcoded password.
[Here](https://github.com/reinhard-brandstaedter/solarflow-control/issues/204#issuecomment-2351493677) is how you can get the password for your hub's id.

I-H-deBoer, 2025-02-02:
user/pwd is on mosquitto. But also with enabled anonymous login, the behavior does not change as mentioned in the original post.
I did not change anything on the hub. 

Before running the script, I set
SET WIFI_PWD=<pwd>
SET SF_DEVICE_ID=<id>
SET SF_PRODUCT_ID=73bkTV
SET MQTT_USER=<user>
SET MQTT_PWD=<pwd>

That's it. Hub has latest Firmware 2.0.48

Do I need to set the hub's pwd somewhere?

reinhard-brandstaedter, 2025-02-04:
Hm. You can’t set the hub’s password - it’s hardcoded as described.
All the cases with such a behavior were rooted in failed authentication on the MQTT side.
Did you really afd a mqtt user with the correct password of the hub (check upper/lowercase).

Last resort would be a tcpdump on your mqtt broker’s interface.

derron7, 2025-02-04:
Nearly the same issue was solved in my case by publishing the following MQTT statement:

`iot/{SF_PRODUCT_ID}/{SF_DEVICE_ID}/properties/read {"properties": ["getAll"]}`

I-H-deBoer, 2025-02-04:
Thank you both.  Some feedback on that.

> Hm. You can’t set the hub’s password - it’s hardcoded as described. All the cases with such a behavior were rooted in failed authentication on the MQTT side. Did you really afd a mqtt user with the correct password of the hub (check upper/lowercase).

Understood about the hardcoded password. I thought I need to put it somewhere in some py-script. But I think this is not the problem.

I checked again on my mosquito broker... in the config I put "allow_anonymous true". To test it, I removed user/pwd from my shelly and opendtu. Both can connect without user/pwd and I see that in my MQTT explorer.

> Last resort would be a tcpdump on your mqtt broker’s interface.

I will check on this asap.

> `iot/{SF_PRODUCT_ID}/{SF_DEVICE_ID}/properties/read {"properties": ["getAll"]}`

Thanks for that. I am trying to catch up on MQTT and not such an expert up to now. Should I put that statement into the py-script? like after the connection?

```
            if connect and ssid:
                await set_IoT_Url(bt_client,"mq.zen-iot.com",1883,ssid,SF_DEVICE_ID)
		mq_client.publish(f"iot/{SF_PRODUCT_ID}/{SF_DEVICE_ID}/properties/read {"properties": ["getAll"]}")
                log.info("Setting IoTURL connection parameters - connect")
                await asyncio.sleep(30)
                return
```

derron7, 2025-02-04:
If you are already receiving the register messages, you are almost there. In my opinion you only need to publish the getall message. This is easily done via the MQTT Explorer. For example https://community.symcon.de/t/zendure-smartflow-mqtt/134538/46 but use the correct json string: 
`{"properties": ["getAll"]}` 
and replace SF_PRODUCT_ID und SF_DEVICE_ID with your IDs.

I-H-deBoer, 2025-02-06:
Thanks for the hint. My proposal was wrong anyhow as I proposed it in the wrong place.

Checked the forum link. Executed the MQTT statement in MQTT explorer. Did not help. I have the same image as mentioned in the forum. I now see register and read entry, but nothing happens.

`python solarflow-bt-manager.py -i -b broker-ip:1883`

works without problems, but not the normal disconnect.

I-H-deBoer, 2025-02-07:
I will close this issue. No solution insight. Also, all tips from other forum did not help to take my system offline. Thanks for your support.

closed 2025-02-07 by I-H-deBoer