reinhard-brandstaedter/solarflow-bt-manager #37: No Solarflow device found!

der-picknicker, 2025-02-25:
Hi @ all,

I started up my balcony power station a few days ago (a Hyper 2000 with two AB2000s)...
At first I only connected it to the Zendure app via Bluetooth - that worked so far.
Then I wanted to establish a connection with the solarfow-bt-manager to configure the hub so that it sends the data to my Homeassistant installation and the MQTT running in it...
Unfortunately, I only got the message “INFO: No Solarflow device found! You can try...”

I then tried all the suggested solutions several times, unfortunately without success.
Then I thought maybe I had to connect to the WLAN and thus the Zendure Cloud via the app first - that didn't help either :(

I have now tested this again with a second notebook (the first has Windows 10 on it, the second Linux Mint, both show the hub as “ZenE1_28 ” in the Bluetooth settings)...

Any ideas what else I can do / test?

reinhard-brandstaedter, 2025-02-26:
Hi @der-picknicker ,
The Hyper2000 gave us some troubles regarding product ID in the past.
Can you try again with the latest version of the script. Eventually also check the reported product ID of your device.
But generally it should be detected.

kr,
Reinhard

der-picknicker, 2025-03-01:
Hi Reinhard,
I tried this out again today...
Unfortunately, I still can't establish a connection 👎 
The Zenure app says ID: 69020
Could it have something to do with that?
The ID should actually be "ja72U0ha", right?

best regards
Philipp
alias der-picknicker

der-picknicker, 2025-03-02:
Hi, me again (maybe don't close the topic yet)...

As already written, I was able to connect...
However, the Solarflow cannot send the data to the MQTT in the HA - the following message always appears in the log:
“Client bgt6Uqk0 disconnected, not authorized.”

Apart from that, I have assigned a different mqtt user via the script - but the Hyper 2000 does not seem to accept this.

In the Homeassistant, however, I cannot enter “bgt6Uqk0” as a user either (username_not_normalized appears because of the capital “U” - with a small “u” I can create the user, but the message remains the same)!

I have read that the MQTT broker must allow anonymous login and also that this obviously does not work with the Mosquitto in Homeassistant...
But I read somewhere else that it is possible somehow (also that you should probably create the ID as a user) - unfortunately I can no longer find the post :(

Maybe someone here can still help me?

kind regards
Philipp
alias der-picknicker :)

der-picknicker, 2025-03-04:
Hi,
In the meantime I got the hint I was looking for from dfigus on how to determine the MQTT user name and password of the Hyper 2000 - now the Hyper connects to MQTT - but unfortunately I still don't get a date :(
I have already described the problem here:
https://github.com/dfigus/addon-solarflow-control/issues/57

dfigus, 2025-03-05:
Let's keep the discussion together instead of splitting it between different repos and issues. As far as I understood in dfigus/addon-solarflow-control#57 you could successfully connect the hyper to your local broker and also the solarflow-control. As the Zendure devices report the data a bit strange, solarflow-control periodically triggers the data report and also unpacks the data into different MQTT topics for easier consumption (also providing HA templates). Hence I would propose to the a few checks to figure out where it's broken on your end:

1. Check whether the Hyper is connected and reports values to your MQTT broker: Connect via MQTT Explorer (or any other tool of your choice) and check for `/<productId>/<deviceId>` topics. It seems that productId of Hyper2000 is `ja72U0ha` and your deviceId `bgt6Uqk0`. If you don't see anything there, probably your Hyper is not connected to your local MQTT broker and you need to run solarflow-bt-manager again.
2. Check whether `solarflow-hub/<deviceId>/control` topics and `homeassistant/sensor/solarflow-hub-*` are there. If not you probably also configured the addon/solarflow-control with incorrect MQTT credentials.
3. Check whether you configured the addon/solarflow-control correctly with the above mentioned `productId` and `deviceId`. This will generate topics `iot/<productId>/<deviceId>/properties/read`. The ids need to match the ones from 1.

der-picknicker, 2025-03-05:
Yes, exactly, that's roughly the status of my setup...

Since I was already thinking something like that, I installed an MQTT Explorer addon yesterday and have already had a look there...
I saw a lot of “solarflow entries”, but firstly I couldn't really do anything with them and secondly, as far as I could see, all the entries had no specific values!

But I have the feeling we're getting closer :)
But one after the other (Answers to your points):
1. There is no such entry! So I mean /ja72U0ha/bgt6Uqk0 - I have executed the solarflow-bt-manager script at least once on each of the last two days, but unfortunately obviously without success
2. This entry exists and values are also there
3. This entry also exists, but logically without values, since the entries from 1. are also missing and cannot be transferred...

To point 1 again or to the solarflow-bt-manager script two things:
First: As already written in this thread above, I had to write the user name in the MQTT Broker (i.e. in Homeassistant) with a **small "u"** although the device ID comes with a capital “U” (but a capital “U” was not accepted!), - could this be the reason?
According to the MQTT protocol, however, a connection is established!
Second: As already written in the other thread (https://github.com/dfigus/addon-solarflow-control/issues/57)
The only thing I notice in the log is the difference that the Hyper 2000 reports the following:
At the first position in the bracket appears **p1**:
`New client connected from x.x.x.x:... as bgt6Uqk0 (p1, c1, k60, u'bgt6Uqk0')`
All other devices have **p2** at the first position in the brackets...
For example, from a few socket adapter plugs:
`New client connected from x.x.x.y:... as DVES_6959DF (p2, c1, k30, u'tasmota')`
All other devices (including your solarflow-addon) also deliver a **p2** at this point!
Could it perhaps have something to do with that?
But I have not yet found out what this means.

I am really glad that you are so patient here and are still trying to help - many, many thanks!!!
I'm on the verge of despair - so far the integrations in Homeassistant have always worked so well and easily - I didn't expect it to be such a big deal with the Hyper 2000!

Best regards
Philipp

dfigus, 2025-03-06:
So it looks like your Hyper is not properly connected to your MQTT broker. I have no clue what the p1 and p2 means, you need to check with the manuals of our MQTT broker - also it's strange that you can't use upper case user names. It seems that you are using Mosquitto with HA user - maybe you should try to create a local Mosquitto user with the proper user name as described in the [Mosquitto Add-On docs](https://github.com/home-assistant/addons/blob/master/mosquitto/DOCS.md#option-logins-optional).

der-picknicker, 2025-03-06:
Hi,
I have implemented your tip, so I deleted the Homeassistant user and added a user directly in the MQTT configuration - the big “U” then also works! But unfortunately without success...
The same message appears in the log as before (New client connected from x.x.x.x:... as bgt6Uqk0 (p1, c1, k60, u'bgt6Uqk0')).

But I also reconnected the hyper with the script (two/three times) and accidentally saw in the MQTT Explorer that the desired entry (/ja72U0ha/bgt6Uqk0) appears shortly after executing the script (first screenshot), but disappears again after a short time (second screenshot - I here expanded all relevant entries, maybe that will help with the analysis)!

[image]
[image]

dfigus, 2025-03-07:
So your Hyper is connected, but in my opinion you incorrectly configured solarflow-control via my addon. The Hyper reports with productId `gDa3tb` (topic `/gDa3tb/bgt6Uqk0/` from first screenshot), but solarflow-control requests the data with `ja72U0ha` (topic `iot/ja72U0ha/bgt6Uqk0` from second screenshot). So this would have been check number 3 from post above:

> 3. Check whether you configured the addon/solarflow-control correctly with the above mentioned `productId` and `deviceId`. This will generate topics `iot/<productId>/<deviceId>/properties/read`. The ids need to match the ones from 1.

Can you try to update the configuration from the addon with product id = `gDa3tb`? This is how it looks on my end (with the value for Hub 2k):

[image]

I can update the description with the values for the hyper to make it easier.

@reinhard-brandstaedter It looks like the product id of the hyper 2000 in solarflow-bt-manager is incorrect and differs from the one in solarflow-control:
https://github.com/reinhard-brandstaedter/solarflow-bt-manager/blob/7c25dfb0ce819d153af0c8b5d0d09d9747008fc0/src/solarflow-bt-manager.py#L140

vs.

https://github.com/reinhard-brandstaedter/solarflow-control/blob/2d9ed3484ec70b681c407645a13a0fba59762a35/src/config.ini#L15

der-picknicker, 2025-03-07:
Ok, shit, I overlooked the different product IDs👎 
But now it has been corrected:
Changing the product id in the addon configuration alone did not work (even reinstalling the addon was not successful)!
Only when I ran the bt script again did the new iot entry appear in MQTT...

[image]

However, the HA is still not receiving any data:

[image]

Have I already mentioned that I'm slowly going crazy

dfigus, 2025-03-07:
This is how it should look like, the device ID is masked (but everywhere the same value): 

[image]

After some time you see in the addon log the following line:
```
Triggering telemetry update: iot/A8yh63/9sH7A2Y5/properties/read
```
And this should then result in the last topic and message from the above screenshot. 

For further analysis please provide the addon configuration and the logs. I still think that there is some wrong configuration on your end.

der-picknicker, 2025-03-08:
Ok, the trigger comes regularly, but there is no data in the addon (log below) and therefore no telemetry data appears in the MQTT...

Here ist the addon config:

> dtu_type: OpenDTU
> smartmeter_type: Smartmeter
> full_charge_interval: 32
> control_bypass: true
> control_soc: false
> battery_low: 10
> battery_high: 100
> mqtt_host: homeassistant
> mqtt_user: zendure
> mqtt_password: [secret]
> dtu_base_topic: solar
> inverter_max_power: 2000
> opendtu_inverter_serial: ""
> ahoydtu_inverter_id: "0"
> ahoydtu_inverter_name: ""
> sm_generic_base_topic: ""
> sm_generic_cur_accessor: ""
> sm_generic_total_accessor: ""
> sm_rapid_change_diff: 500
> sm_zero_offset: 20
> sm_scaling_factor: 1
> sm_shelly3em_base_topic: shellies/shellyem3/
> min_charge_power: 125
> max_discharge_power: 150
> limit_inverter: true
> max_inverter_limit: 800
> max_inverter_input: 700
> min_inverter_limit: 10
> discharge_during_daytime: false
> sunrise_offset: 60
> sunset_offset: 60
> product_id: gDa3tb
> device_id: bgt6Uqk0
> sf_inverter_channels: 1,3

an here's the addon log:

> 2025-03-08 12:08:05,467:INFO: Triggering telemetry update: iot/gDa3tb/bgt6Uqk0/properties/read
> 2025-03-08 12:09:05,365:INFO: HUB: S:-1.0W [ ], B: -1% (-1) low: -1 high: -1, V:-1.0V (-1.0), C: 0W, P:False (manual, possible), F:-1.0h, E:-1.0h, CT:OFF (32hrs) idle, H: -1W, L: -1W
> 2025-03-08 12:09:05,365:INFO: INV: AC:0.0W, DC:0.0W (), L: 0W (0.0W/channel) [ -1W]
> 2025-03-08 12:09:05,365:INFO: SMT: T:Smartmeter P:0.0W [ ]
> 2025-03-08 12:09:05,467:INFO: Triggering telemetry update: iot/gDa3tb/bgt6Uqk0/properties/read
> 

I have also just discovered the following lines:

> 2025-03-08 12:11:05,177:INFO: Publishing Homeassistant templates...
> 2025-03-08 12:11:05,221:INFO: Published 36 Homeassistant templates.

dfigus, 2025-03-11:
> [image]

In this screenshot the battery level (`/gDa3tb/bgt6Uqk0properties/report` with `electricLevel`) was reported with 15%, so the hyper seems to be connected. Also the product and device ID seem to match now. Your configuration has some other issues (such as smartmeter configuration missing and you don't have a DTU connected at all) but this should not harm the data forwarding in my opinion. Keep the MQTT explorer open for a longer time to see if more data is reported there and whether you see the message in `iot/gDa3tb/bgt6Uqk0/properties/read` from `solarflow-control`.

Could you paste the complete log file of the addon? The startup is truncated and hence is missing essential information such as the topic subscription. Press the download log button at the top of the addon log tab.

Also please check whether new HA entities are created as you changed the device ID and hence I would assume this leads to new entities (as the device ID is part of the unique ID).

PS: I think this is rather a solarflow-control configuration issue and no longer a solarflow-bt-manager one.

der-picknicker, 2025-03-14:
Hi, I've been on a business trip this week, so I'm only now getting around to looking again...

To the point:
1. attached is the complete log file...
2. in the MQTT Explorer I have not seen any entry `iot/gDa3tb/bgt6Uqk0/properties/read`!
(how long can that take?)
3. the entities do exist, but from my point of view there are no “new” entries, because the entities do not contain an ID...
An example would be: `sensor.solarflow_hub_solar_input_power`
4. As you write, it is obviously no longer a problem with the solarflow-bt-manager.
Should we therefore close this thread and open a new one under your addon?

Best wishes and a good start to the weekend :o)
Philipp

[dae55a80_solarflow-control_2025-03-14T19-50-54.895Z.log](https://github.com/user-attachments/files/19254011/dae55a80_solarflow-control_2025-03-14T19-50-54.895Z.log)

dfigus, 2025-03-21:
In the log you clearly see an error message that the smartmeter is incorrectly configured. Please configure the smartmeter correctly as outlined in the solarflow-control docs. I don't know whether the incorrect configuration has some side effects.

In the log there are multiple entries for `iot/gDa3tb/bgt6Uqk0/properties/read`, so if you don't see them in MQTT explorer, are you sure you connected both the solarflow-control and the hyper to the same MQTT broker?

Also in the logs the time is after sunset and in case the battery is empty the hyper disconnects.

der-picknicker, 2025-03-23:
> In the log you clearly see an error message that the smartmeter is incorrectly configured. Please configure the smartmeter correctly as outlined in the solarflow-control docs. I don't know whether the incorrect configuration has some side effects.

That's it - smartmeter config done in the addon and tata, the data is there!

[image]

I would not have thought that this could be the reason, since the samrtmeter has nothing to do with the data of the Solarflow...
But never mind - the main thing is that it works now!

But now I actually have another question:
I'm planning to buy a second PV system for the garage - so the question is whether the add-on can also be used to integrate two Hyper 2000s in parallel in HA, for example?
I would also like to discuss this elsewhere - tell me if I should make a new post for this, and if so, where...
Or if there is a discussion forum where you can discuss this.

Best wishes and have a nice rest of Sunday

closed 2025-03-23 by der-picknicker