reinhard-brandstaedter/solarflow-bt-manager #39: SolarFlow 800

thiloplanz, 2025-03-06:
Can support for the new SolarFlow 800 be added?

Does that still use the same basic communication methods as previous models?

(Looking at https://github.com/dav1dBoy/zenSDK it appears that they have added an embedded HTTP API server to the device itself that announces itself over mDNS, which presumably simplifies things)

reinhard-brandstaedter, 2025-03-07:
Hi @thiloplanz,
thanks for pointing me to that.
As an independent developer with no affiliation to Zendure it is hard to keep up with their ever new models. I don't have any access to the SF800 to play with it.
It's interesting however to see that there are even Zendure engineers who are now exposing some internal stuff publically. I can reach out to David and see what information we can exchange.

Meanwhile I can only encourage you to "just try it" and see what happens ;-)
I'd assume the SF800 still reports telemetry via MQTT so the Zendura App can still be used the same way.
The bt-manager will not work out of the box as the SF800 has a new product ID which needs to be found out first.

Report back 

kR

flattman, 2025-03-24:
Hi,
I got my SolarFlow 800 one hour ago and played around with it a bit.

```
SF_PRODUCT_ID = "B1NHMC"
product_class = "zenw"
```

hope this is correct, graped the product ID from the first 6 chars of the bluethooth ManufacturerData.Value. The device info command returns some data but maybe there is a little more to do.

```
2025-03-24 12:55:40,098:INFO: {'messageId': 9, 'timestamp': 1764, 'deviceId': 'xxxx', 'sn': 'xxxx', 'method': 'getInfo-rsp', 'modules': [{'module': 'MASTER', 'version': 4115}, {'module': 'AC', 'version': 4361}, {'module': 'BMS', 'version': -1}, {'module': 'BMS_AB2000', 'version': -1}, {'module': 'ESP32', 'version': 4352}]}
```

had only one error:

```
2025-03-24 12:55:40,099:ERROR: A message handler raised an exception: 'deviceSn'
Traceback (most recent call last):
  File "src/dbus_fast/message_bus.py", line 802, in dbus_fast.message_bus.BaseMessageBus._process_message
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/manager.py", line 1008, in _parse_msg
    watcher.on_characteristic_value_changed(
  File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/client.py", line 177, in on_value_changed
    callback(bytearray(value))
  File "/opt/devserver/stuff/solarflow-bt-manager/src/solarflow-bt-manager.py", line 106, in handle_rx
    log.info(f'The SF device SN is: {payload["deviceSn"]}')
```

but i got this messages

```
2025-03-24 12:55:50,674:INFO: {'timestamp': 1774, 'messageId': 10, 'method': 'report', 'deviceId': 'xxxx', 'properties': {'heatState': 0, 'packInputPower': 0, 'outputPackPower': 0, 'outputHomePower': 0, 'remainOutTime': 59940, 'packState': 0, 'packNum': 0, 'electricLevel': 0, 'gridInputPower': 0, 'solarInputPower': 0, 'solarPower1': 0, 'solarPower2': 0, 'pass': 0, 'reverseState': 0, 'socStatus': 0, 'hyperTmp': 2981, 'dcStatus': 0, 'pvStatus': 0, 'acStatus': 0, 'dataReady': 1, 'gridState': 1, 'BatVolt': 6}}
2025-03-24 12:55:50,823:INFO: {'timestamp': 1774, 'messageId': 10, 'method': 'report', 'deviceId': 'xxxx', 'properties': {'writeRsp': 0, 'acMode': 2, 'inputLimit': 800, 'outputLimit': 200, 'socSet': 1000, 'minSoc': 50, 'gridStandard': 0, 'gridReverse': 1, 'inverseMaxPower': 800, 'lampSwitch': 1, 'IOTState': 1, 'factoryModeState': 0, 'OTAState': 0, 'LCNState': 0, 'oldMode': 0, 'VoltWakeup': 0, 'ts': 1742817348, 'bindstate': 0, 'tsZone': 0, 'chargeMaxLimit': 800}}
```

At the moment I can not reconfigure the network and set a local mqtt server. I will investigate some time later... If anyone needs more infos, let me know

thiloplanz, 2025-03-31:
I also got my SolarFlow 800 last week.

So far, I have not been able to connect to it using solarflow-bt-manager, but that may be an error on my end (or some problem with the Mac hardware I am using), I am really not knowledgable enough about Bluetooth.

@flattman are there any debug steps or hints or caveats to the procedure? Did it "just work" for you? 

For now, I wrote a small script that calls their new HTTP API on the local network and copies the results into MQTT in the same way that solarflow-topic-mapper would do. This way I got solarflow-statuspage to work. Bit too round-about for my taste, but it was fun. Next step is trying to get HomeAssistant to discover the sensors via MQTT (by emulating the configs from https://github.com/reinhard-brandstaedter/solarflow-control/tree/master/src/solarflow/homeassistant )

xammmue, 2025-04-07:
> Hi, I got my SolarFlow 800 one hour ago and played around with it a bit.
> 
> 
> had only one error:
> 
> ```
> 2025-03-24 12:55:40,099:ERROR: A message handler raised an exception: 'deviceSn'
> Traceback (most recent call last):
>   File "src/dbus_fast/message_bus.py", line 802, in dbus_fast.message_bus.BaseMessageBus._process_message
>   File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/manager.py", line 1008, in _parse_msg
>     watcher.on_characteristic_value_changed(
>   File "/usr/lib/python3/dist-packages/bleak/backends/bluezdbus/client.py", line 177, in on_value_changed
>     callback(bytearray(value))
>   File "/opt/devserver/stuff/solarflow-bt-manager/src/solarflow-bt-manager.py", line 106, in handle_rx
>     log.info(f'The SF device SN is: {payload["deviceSn"]}')
> ```
> 
> but i got this messages
> 
> ```
> 2025-03-24 12:55:50,674:INFO: {'timestamp': 1774, 'messageId': 10, 'method': 'report', 'deviceId': 'xxxx', 'properties': {'heatState': 0, 'packInputPower': 0, 'outputPackPower': 0, 'outputHomePower': 0, 'remainOutTime': 59940, 'packState': 0, 'packNum': 0, 'electricLevel': 0, 'gridInputPower': 0, 'solarInputPower': 0, 'solarPower1': 0, 'solarPower2': 0, 'pass': 0, 'reverseState': 0, 'socStatus': 0, 'hyperTmp': 2981, 'dcStatus': 0, 'pvStatus': 0, 'acStatus': 0, 'dataReady': 1, 'gridState': 1, 'BatVolt': 6}}
> 2025-03-24 12:55:50,823:INFO: {'timestamp': 1774, 'messageId': 10, 'method': 'report', 'deviceId': 'xxxx', 'properties': {'writeRsp': 0, 'acMode': 2, 'inputLimit': 800, 'outputLimit': 200, 'socSet': 1000, 'minSoc': 50, 'gridStandard': 0, 'gridReverse': 1, 'inverseMaxPower': 800, 'lampSwitch': 1, 'IOTState': 1, 'factoryModeState': 0, 'OTAState': 0, 'LCNState': 0, 'oldMode': 0, 'VoltWakeup': 0, 'ts': 1742817348, 'bindstate': 0, 'tsZone': 0, 'chargeMaxLimit': 800}}
> ```
 
Hi,
I was just looking into this project to figure out whether it would be possible to add a 2400 AC, which doesn't seem possible, but maybe I can help you out a bit.

The line of code that raised that exception you got isn't strictly required, so you could maybe get the whole thing to work if you just deleted it.
I am not familiar with the project (yet), but this should at least lead to some messages being published to mqtt that contain the properties from your other logs.

The line in question is line 106 in solarflow-bt-manager.py
```
log.info(f'The SF device SN is: {payload["deviceSn"]}')
```
Either remove it or disable it by adding a `#` in front of `log`

flattman, 2025-04-11:
Hi @thiloplanz,

Sorry for the late reply.

Here are my changes: https://github.com/flattman/solarflow-bt-manager/tree/feature/support-solarflow-800

However, I've now switched to the REST API and retrieve the data from the inverter locally and pull it into my Home Assistant. That is enough for me. I initially played around with setting my MQTT broker, but then gave up without success and changed to the REST API.

thiloplanz, 2025-05-26:
> I take it you are talking about the solarflow 800 pro battery + inverter not the micro inverter solarflow 800, right?

@joba-1 

Both the Solarflow 800 and the 800 Pro support this new local HTTP API, and also presumably still the old BlueTooth connection flow (at least it can be used in their official app, I have not really tried to use it with solarflow-bt-manager), so you can absolutely monitor and control them fully locally without a cloud connection.

thiloplanz, 2025-07-18:
> Looking at https://github.com/dav1dBoy/zenSDK it appears that they have added an embedded HTTP API server to the device itself that announces itself over mDNS, which presumably simplifies things

Hmm... That repository has been deleted. 
I hope they are not planning to shut the API down in a future firmware update (probably not, as the Zendure homepage mentions "local control" a lot now).

thiloplanz, 2025-07-31:
The local HTTP API is now documented on Zendure's official Github repo: https://github.com/Zendure/zenSDK