reinhard-brandstaedter/solarflow-bt-manager #42: Issue with Mosquitto broker

juliuscorsus, 2025-03-16:
Hi all, 

I have a Zendure AB1200 with two batteries. Currently, I'm using Home Assistant to read values directly from the Zendure MQTT broker, and it works fine. However, I would like to have more control over the generated power from Solarflow by implementing specific algorithms that Solarflow cannot accommodate.

I came across the "Solarflow Control" and "Solarflow BT Manager" tools. I was able to set everything up by following the instructions step by step, but I'm having trouble getting it to work. Actually I was able to disconnect from cloud.  I suspect that the Mosquitto broker in Home Assistant might be the issue because the documentation states: "Unfortunately, it will use the hub's default (hardcoded?) password. So, for now, you will need to set your MQTT broker to accept any login (remove authentication)."
However, Mosquitto broker removed the option to allow anonymous authentication in 2022.

All the properties in HA are unknown and in MQTT Explorer I have only "solar-flow hub" and "iot" in my local MQTT broker but not all the required information. 

Any idea on how to solve?
[Im planning in near future to setup a PC with proxmox and HA. IT could be an idea to setup a parallel MQTT broker in proxmox as that allows anonymous authentication. There is an alternative?]
Many Thanks

reinhard-brandstaedter, 2025-03-17:
Hi @juliuscorsus,
The password can be retrieved from the hub’s ID:

See:
https://github.com/reinhard-brandstaedter/solarflow-control/issues/204#issuecomment-2351493677

(watch out for lower/upper case).
If you create that user/pwd on the mosquitto side (user is the hub ID) you should be able to connect.

juliuscorsus, 2025-03-17:
Unfortunately, I haven't had any luck. I'm using the device ID and password on Mosquitto. Currently, I see the product ID item in MQTT Explorer, but it only shows a single item that doesn't change over time. Additionally, in Solarflow Control, everything still appears as "unknown."

What could be the issue?

Thank you very much for your support.

juliuscorsus, 2025-03-18:
Hi all, 
I tried using device ID an password on Mosquitto broker, but still no luck. 

Here the setup through bt-manager:
[image]
It looks correct. The cloud is actually disabled on the app. 

But, this is the only I can get from MQTT explorer:
[image]

Here the log of Mosquitto broker:
[image]

Anyone is able to spot the error?
Many thanks

juliuscorsus, 2025-03-18:
As update, I have done a step ahead:
if I use the option -i when disconnecting from the cloud, I can see all the telemetry data from MQTT explorer. 
But I guess the infos are taken through BT and not wifi to the mqtt broker. 
If I play through the HA Solarflow control, I see the command on MQTT explorer, but no real changes. 
All the sensors in HA are still "unknown". 

I see that im getting closer to the solution, but still...

juliuscorsus, 2025-03-19:
This message is to confirm that I finally succeeded and to suggest a possible solution for anyone who has experienced the same issue: the Hub was not sending information to the local MQTT broker because I had chosen the SSID of the WiFi connection of the WiFi repeater instead of the main router. For some reason, this was blocking everything and preventing the information from reaching its destination.
Now it seems to be working.

Thanks again to @reinhard-brandstaedter for developing everything.

reinhard-brandstaedter, 2025-06-03:
Glad you figured it out @juliuscorsus !