reinhard-brandstaedter/solarflow-bt-manager #47: Hub 2000 is not establishing a connection to my wifi

CWILDMOUNTAIN, 2025-04-21:
Hi everyone

after over a week of trying to get my Hub 2000 reporting to my local mqtt-broker over Wifi I've no clue what the problem is and how to solve it.
So let me introduce you to my setup: 

a) homeassistant green; this is where there mqtt-broker is running (I'm using EMQX as broker and had no issues with anonymous login at any time) and the topic mapper as well; it has no BT though

b) MacBook with python 3; I run the bt-manager in terminal from this platform whereas the homeassistant green has noch BT; of course both are connected to the same wifi

c) access point is a Fritz!Box with WPA 2 (connection to the cloud can be established over this network)

so here is my procedure and my problem respectively:

1. connected the Hub only via BT on the phone, not to Zendure server over wifi (tried this as well, but same problem) 
2. switch off BT on my phone
3. now I run the script with the -i -b option and I can see values population in my mqtt explorer in homeassistant; of course the IP-adress shown is now the IP-Adress from my MacBook, whereas the data is just pushed over the laptops BT to Wifi (if I understand correctly)
4. If I did the same now with the -d -w option, the log sais disconnect, but there is no connection established of the hub to my wifi; I think I tried everything (disabled BT on the Laptop after disconnect, restart the hub pushing the iot-button, even tried different pushing-durations (3s, 6s, 10s+) and also tried to disconnection the Hub from panels and batteries completely)

So long story short- no connection to my wifi, IOT button is flashing slowly and the device is noch showing up in my network and I'm not able to ping; so I assume: the Hub my not be connected to the cloud but is also not connected to my wifi at all.
I tried a different wifi as well- same story.

Has anyone faced similar problems or has any helpful hint, what I'm doing wrong here?
Thanks in advance!

reinhard-brandstaedter, 2025-04-21:
Did you specify the WiFi password correctly in the environment variable? Does it maybe contain any special characters the the shell command could interpret wrong?
Try to switch the hub to a different temporary Wifi and then back to the final one.

CWILDMOUNTAIN, 2025-04-21:
the only special character is: - (minus)
well - as I said, I tried to connect to a different WiFi, also changed the WiFi PW .. no connection in any case

CWILDMOUNTAIN, 2025-04-21:
Thank you and anybody else for your hard work, I found a solution I want to share with anybody else having trouble and may not be as good as you guys in programming stuff. 

For whom it might be ok, not work completely offline but get control in homeassistant: this looks really interesting and promising: 
https://github.com/FireSon/Zendure-HA

closed 2025-04-21 by CWILDMOUNTAIN