reinhard-brandstaedter/solarflow-bt-manager #51: Error while trying to run the script

xBorsti, 2025-06-03:
Hi, 

i am on a fresh ubuntu install with python3 installed and i also have installed the requirements from the requirements.txt. 

If i run the script with -i i get the following error:

`2025-06-03 07:55:03,480:INFO: scan for: zenp
Traceback (most recent call last):
  File "/home/xborsti/zendure/solarflow-bt-manager.py", line 262, in <module>
    main(sys.argv[1:])
  File "/home/xborsti/zendure/solarflow-bt-manager.py", line 259, in main
    asyncio.run(run(broker=mqtt_broker, port=mqtt_port, info_only=info_only, connect=connect, disconnect=disconnect, ssid=ssid))
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/xborsti/zendure/solarflow-bt-manager.py", line 153, in run
    device = await BleakScanner.find_device_by_filter(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/xborsti/python3/lib/python3.12/site-packages/bleak/__init__.py", line 444, in find_device_by_filter
    async with cls(**kwargs) as scanner:
  File "/home/xborsti/python3/lib/python3.12/site-packages/bleak/__init__.py", line 158, in __aenter__
    await self._backend.start()
  File "/home/xborsti/python3/lib/python3.12/site-packages/bleak/backends/bluezdbus/scanner.py", line 168, in start
    manager = await get_global_bluez_manager()
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/xborsti/python3/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 1058, in get_global_bluez_manager
    await instance.async_init()
  File "/home/xborsti/python3/lib/python3.12/site-packages/bleak/backends/bluezdbus/manager.py", line 286, in async_init
    assert_reply(reply)
  File "/home/xborsti/python3/lib/python3.12/site-packages/bleak/backends/bluezdbus/utils.py", line 20, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.ServiceUnknown] The name org.bluez was not provided by any .service files
`

i tried to install the asyncio-package also, but that didnt change anything. 

can someone help me with that?

reinhard-brandstaedter, 2025-06-03:
Is the bluetooth/bluez stack installed on your system?
```
apt install bluetooth bluez
```

xBorsti, 2025-06-03:
it seems, that was the problem thanks.

closed 2025-06-03 by xBorsti