reinhard-brandstaedter/solarflow-bt-manager #53: Connection drops after receiving payload on Solarflow 2400 AC

account-closed, 2025-06-21:
## : Solarflow 2400 AC Drops Connection After Sending Payload

### Description
When sending a valid configuration payload to the Solarflow 2400 AC (**Product ID: BC8B7F**) via Bluetooth, the device accepts the payload but immediately drops the BT  connection and exits setup mode (the blinking Wi-Fi symbol stops). No MQTT connection is attempted from the device afterward.

### Environment
- **Device:** Solarflow 2400 AC
- **Product ID:** BC8B7F 
- **Firmware Version:** v1.0.4 
- **Host OS:** Linux 6.8 
- **Python Version:** 3.12 

### Steps to Reproduce
1. Put the device into setup mode (blinking Wi-Fi symbol).
2. Run the following command:
   ```
   python solarflow-bt-manager.py -d -w <SSID> -b <MQTT_BROKER>
   ```
   *(Ensure required environment variables are set.)*
3. Observe the Bluetooth communication and device behavior.

### Expected Behavior
The device should:
- Accept the payload,
- Connect to the specified MQTT broker,

### Actual Behavior
- The payload is accepted silently vi BT without response.
- The Bluetooth connection is immediately dropped.
- The device leaves setup mode, but **never attempts to connect to the MQTT broker**.

### Console Output (redacted)
```
Disconnecting Solarflow Hub from Zendure Cloud  
2025-06-21 17:20:05,641:INFO: scan for: zen  
2025-06-21 17:20:05,838:INFO: Found device: <MAC_ADDRESS>: ZenHA1_2  
2025-06-21 17:20:06,847:INFO: Services:  
2025-06-21 17:20:06,847:INFO: 0000a002-0000-1000-8000-00805f9b34fb (Handle: 40): Vendor specific  
2025-06-21 17:20:06,847:INFO: 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile  
2025-06-21 17:20:06,899:INFO: {"iotUrl": "<MQTT_BROKER>", "messageId": "1002", "method": "token", "password": "<REDACTED>", "ssid": "<SSID>", "timeZone": "GMT+02:00", "token": "abcdefgh"}  
2025-06-21 17:20:06,900:INFO: Setting IoTURL connection parameters – disconnect  
```

### Additional Context & Troubleshooting

- ✅ No connection attempts seen from the device on the network to the broker (confirmed via packet dump and AP logs).
- ✅ Tried multiple MQTT brokers and SSIDs – same result.
- ✅ Factory reset and reboot did not help.
- ✅ Wifi config is ther probably from the initial config 
- ❓ No further feedback or status from the device after the disconnection.

---

Would logs, BLE packet dumps, or additional debug info be helpful? Happy to provide more details.

heisly, 2025-08-16:
Any progress using solar flow 2400 offline via local MQTT?

STEF0112, 2025-09-22:
My new SF 2400 AC was deliverd now.
Can I control this with solarflow-bt-manager also ?

reinhard-brandstaedter, 2025-11-23:
I do not have access to a SF 2400 AC, so need to rely on the community here...

maxclaudi-tech, 2026-08-19:
> ## : Solarflow 2400 AC Drops Connection After Sending Payload
> ### Description
> When sending a valid configuration payload to the Solarflow 2400 AC (**Product ID: BC8B7F**) via Bluetooth, the device [...]

There is a working alternative available. You don't need anything except the serial number of the device, your desired broker IP, and your WiFi credentials.

German iobroker-forum: **[zURLtool V 1.0 dch](https://forum.iobroker.net/topic/82263/zendure-zensdk-lokal-api-smartmode-solarflow-ac-800-pro-2/479)** is a new tool that rewrites the Cloud DNS to any desired IP/DNS for both old and new Zendure devices (successfully tested up to the 2400 AC pro).

*Note: The interface of the tool itself is German-only, but the included `ReadMe_zURLtool.txt` is provided in both German and English, and the application is designed to be fully self-explanatory. Screenshots of the GUI can also be viewed in advance in the linked German ioBroker forum thread.*

After the writing process is completed, a file named **`mqtt_creds.txt`** is generated in the same directory with the following content:

```text
MQTT-Broker URL: xxx.xxx.xxx.xxx
MQTT-Broker User: XxXxXxXx
MQTT-Broker Password: XxXxXxXxXxXxXxXx
DeviceId: XxXxXxXx
Zendure Serial Number: XXXXXXXXXXXXXXX
Zendure Product ID: xXxXxX
```
*Note: (Old/Some) Zendure devices do not transmit a Product ID.*

#### How it works & Features:
* The tool scans for Zendure devices via Bluetooth.
* Any found device will list its serial number.
* Compare the detected serial number with your own device: sticker/nameplate or original packaging.
* Correct serial number selected? Then you are ready to write the configuration.
* It is a Windows 11 GUI tool, directly executable by double-clicking `zURLtool_GUI.exe`.
* No installation and no additional setup of runtime libraries etc. are required. Everything is fully implemented inside the tool.
* In the `config.ini`, only the settings for WiFi and the desired broker IP/DNS need to be configured before starting the application.
* You can also write back to the original Cloud DNS at any time.

zURLtool was mainly developed for newer devices (tested up to the AC 2400 pro), but it also works seamlessly with older models.

#### Important Technical Notes:
* The port for the Cloud broker cannot be changed. The Cloud broker port is hardcoded inside the firmware.
* Older devices / older firmware versions still use port 1887 without TLS.
* New firmware versions from the device generation >= Hyper 2000 use port 8883 and enforce TLS—at least regarding the Cloud and Cloud-replacement brokers.